F1ATB/Remote-SDR #2: (Request) Update for python3 / gr 3.8

alphafox02, 2021-06-08:
I’d love to include this in DragonOS. I started looking at the source code and opening the grc files. Appears to be python2/gr 3.7 based. You wouldn’t happen to have 3.8 based flow graphs? Figured I’d ask before working in them.

F1ATB, 2021-06-09:
Hello

As far as I know, the only stable version for Debian Buster is GNU Radio
3.7.13.4-4 (python2)
When the GNU Radio installation package in 3.8 will be available , I will
update Remote SDR to use only Python3.

Regards

alphafox02, 2021-06-09:
You’re right, I was trying to use your code on Ubuntu 20.04. The flow graphs I opened so far only needed minor changes. I was able to make modifications and kind of get it working on x64. Awesome project. I’ll close this ticket since it’s probably irrelevant for the orange pi now.

closed 2021-06-09 by alphafox02