F1ATB/Remote-SDR #3: USB Hub needed?

TheRealMoeder, 2021-06-11:
Hello,

can you explain the issue requiring an USB Hub for the Plutosdr to work with Remote SDR? Did you look into possible solutions?

73

F1ATB, 2021-06-11:
It's an issue not only with Remote SDR. Google it and you will see a lot of
discussion about.
Honestly, I don't know exactly the origin of this bug.

TheRealMoeder, 2021-06-13:
I'm not quite sure which issue exactly you are referring to. But I'll try to look into it as soon as my OrangePi arrives. From what I've read up a lot of changes were made in the past weeks regarding the USB stuff. Which Armbian version is your released image based on?

F1ATB, 2021-06-14:
This morning I redo the test with or without an USB hub. I confirm USB Hub is necessary between the Pluto and the Orange Pi One Plus running Armbian version 21.02.3. In a terminal if you type lsusb, the Pluto is not listed without a hub.
With Orange Pi Zero2 running Debian from Orangepi.org there is no issue.

TheRealMoeder, 2021-06-14:
Yes, Armbian doesn't seem good enough yet, OPi Zero 2 support ist still not in a workable state. I Hope to receive Pi Zero 2 these days and start first tests with Remotesdr. Thanks for the heads up.

closed 2021-06-14 by TheRealMoeder

TheRealMoeder, 2021-06-16:
I can confirm your image works fine without usb hub on the zero 2. Very nice project you created!

F1ATB, 2021-06-16:
Thanks for your positive report.
73