F1ATB/Remote-SDR #4: publish source code in the git repository?

TheRealMoeder, 2021-06-11:
Would you mind publishing the source code directly into the git repository so it can be forked and pull requests made to submit improvements?

73

F1ATB, 2021-06-11:
As I am new in GitHub, I want to simplify the access as I am not confident
for the moment in such management of source code.

André
F1ATB

TheRealMoeder, 2021-06-14:
That is too bad. Try to read up on it, really not that difficult and git makes life a lot easier when developing open source stuff. Someone already forked your early repo but changes cannot bei pulled in without the base repo being setup.

closed 2021-11-08 by F1ATB

mfalkvidd, 2023-03-10:
Sorry for being confused @F1ATB , but where can we download the source code?

F1ATB, 2023-03-10:
https://github.com/F1ATB/Remote-SDR/releases
You will fond releases

mfalkvidd, 2023-03-10:
Great, thanks!