F1ATB/Remote-SDR #6: Enable Direct Sampling for RTL-SDR

jefffall, 2021-09-23:
Need ability to enable Direct Sampling on Q Branch for RTL-SDR V3 to allow good HF sensitivity.

Besoin de pouvoir activer l'échantillonnage direct sur la branche Q pour RTL-SDR V3 pour permettre une bonne sensibilité HF.

F1ATB, 2021-09-23:
Yes, I see. I need to modify the GNURADIO diagram. In a future release of
Remote SDR.
Regards
73
André

jefffall, 2021-09-23:
Thanks you andre. Your software is very nice and good to use. It will be exciting when the HF support is found on rtl_sdr. I could not get a V3 rtl_sdr to work at all with the 2.5 release and so I have returned the rtl_sdr V3 to Amazon. I could not get the rtl_sdr to hear any HF with your remote sdr or with CubicSDR. I am impressed your image the IMG file was very easy to flash to a 64GB SD card and that it did expand to use all the 64GB space. And I was very happy you have the ssh running and that I could log in and su to root and look around. Your efforts are very appreciated.

Merci andré. Votre logiciel est très agréable et bon à utiliser. Ce sera excitant quand le support HF sera trouvé sur rtl_sdr. Je ne pouvais pas du tout faire fonctionner un rtl_sdr V3 avec la version 2.5 et j'ai donc renvoyé le rtl_sdr V3 à Amazon. Je n'ai pas pu faire en sorte que le rtl_sdr entende une HF avec votre sdr distant ou avec CubicSDR. Je suis impressionné par votre image, le fichier IMG était très facile à flasher sur une carte SD de 64 Go et qu'il s'est étendu pour utiliser tout l'espace de 64 Go. Et j'étais très heureux que ssh fonctionne et que je puisse me connecter et su pour rooter et regarder autour de moi. Vos efforts sont très appréciés.

Insidetv, 2021-12-12:
Hello Andre,
I just installed it on a raspberry 4 and it works fine with a RTL-SDT V3 except direct sampling.
Is there any progress regarding this matter?
regards,
Frank

F1ATB, 2021-12-12:
No progress.
It still on my wishes list.
I already look at it. I don't know yet how to switch in direct sampling
with the interface to GnuRadio
73
Andre

closed 2022-02-09 by F1ATB