F1ATB/Remote-SDR #11: CLKOUT back to default (disable) when starting remote sdr

s21rc, 2021-11-06:
I am trying to connect the CLOKOUT of RX hackRF to CLKIN of TX hackRF board. CLOKOUT is off by default in recent hackRF firmware (2021). After enabling the clkout using hackrf_clock utility "hackrf_clock -d xxxx -o 1" when running remotesdr from browser - the clokout gets disabled, I guess because of resetting the radio by remotesdr.

Once remotesdr starts - hackrf_clock do not work as the radio shows busy.

Can you please have a look and put a code to enable hackrf clockout with a #, so people can uncomment it if clkout needed.

Thanks, S21RC

F1ATB, 2021-11-06:
It's the first time, I heard of that issue. I will try to better understand
and investigate what can be done.
Do you have in the RX hackRF a TCXO or do you use the internal clock?

Regards

André

s21rc, 2021-11-06:
Hi Andre,
Here is detail:
The hackRF one board power on as CLKOUT (MS3) disabled. to enable it you have to run: hackrf_clock -d "serial4digit" -o 1

that works fine with both internal clock, and also with TCXO/ and OCXO from CLKIN [by default CLKOUT takes 10MHz from internal clock, the moment you activate the RX/TX then it takes clock from CLKIN/TCXO etc.] This is tested with CUBICSDR in the same Pi.

When firing remoteSDR in a browser, once RX or TX (tested with both RX and TX board) becomes green - the CLKOUT gets disabled. I tried to manually enable the CLKOUT again with hackrf_clock utility - but that gives error that resourse busy, same if I try hackrf_info,

I see that in the Radio start at boot PY file has radio kill and radio start. I tried to put the CLKOUT enable there - no luck.
If I put it before Radio terminate/kill portion, has no effect, if put after radio kill before start - then remSDR fail to turn on the radio. If I put after radio start then get resource busy error. 

BTW: I have upgraded the hackrf /libhackrf and hackrf_info tools to latest from greatscott git.

pi@rasp-rem-sdr:~ $ hackrf_info
hackrf_info version: git-681d0c1
libhackrf version: git-681d0c1 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc24788dc3
hackrf_open() failed: Resource busy (-1000)

Found HackRF
Index: 1
Serial number: 0000000000000000f77c60dc279191c3
hackrf_open() failed: Resource busy (-1000)

Let me know if you can fix this., thanks, 73 Fazlay

s21rc, 2021-11-06:
pi@rasp-rem-sdr:~ $ hackrf_clock -d 8dc3 -o 1
hackrf_open() failed: Resource busy (-1000)

F1ATB, 2021-11-06:
Fazlay I started to look at this issue and discovered that the clock is
disabled at startup for recent hackrf. May be the solution will be at boot
to connect the clock and wait a little to boot remsdr. I will see.

F1ATB, 2021-11-08:
Fazlay
All the HackRF One I have, are chinese version not at the latest firmware as a Great Scott One. It' s difficult for me to do the test. Can you do it for me.?

First I propose to check the command to enable the clock.
- Boot the Raspberry/PI and don't go on Remote SDR.
- Open a terminal and type the command :..$ hackrf_clock -d 8dc3 -o 1

Assuming d8c3 is the end of the serial number.
At this stage you shouldn't have a message Ressource Busy
The purpose is to test that using the last 4 digits of the serial number is sufficient or you need to enter the full serial number given by $ hackrf_info.
If the clock in on, you can go on remote SDR. It should work as usual.

It it's OK, we can enter 1 line of code ** os.system("hackrf_clock -d 8dc3 -o 1") **
Replace in the file start_radio_manager_at boot.py at the beginning :

.......................
sys.stdout = f_log 
sys.stderr = f_log

# copy authorization to apache to access GPIO as root
os.system("cp /var/www/html/cgi-bin/python_web /etc/sudoers.d/python_web")
..........................

by
.......................
sys.stdout = f_log 
sys.stderr = f_log

os.system("hackrf_clock -d 8dc3 -o 1")

# copy authorization to apache to access GPIO as root
os.system("cp /var/www/html/cgi-bin/python_web /etc/sudoers.d/python_web")
..........................

Reboot the system.
Tell me if it works and if you used only 4 digits for the serial number or the complete one.
Regards
73
André

closed 2022-06-04 by F1ATB