F1ATB/Remote-SDR #12: Accessing Remote-SDR via a network translated IP.

Derek000, 2021-11-17:
Hi,

I've been attempting to port forward Remote-SDR from a private network to an Internet address.

Other than port 80, is there any other ports required to be forwarded, can the service be forwarded to non standard ports, and does the internal or external IP address need to be placed in the parameters configuration for the webservice to communicate with the SDR?

I can see that 8006 (RX) and 8007 (TX) can be used by 

I've looked at the documentation and other discussions, but cant seem to find this scenario explained.

Thanks in advance.

F1ATB, 2021-11-17:
You have to put the external IP in the parameters and forward ports 8001 to
8007 to the same internally 8001 to 8007

Regards
73
André

Derek000, 2021-11-17:
Thanks. No I haven't tried that, i'll give it a go.

I would like to utilise some non standard ports. Is there a way to change the default ports that the UI and backend can be changed to?

Thanks again.

F1ATB, 2021-11-17:
For the moment I didn't introduce this flexibility to change the ports.

73
André

Derek000, 2021-11-17:
Thanks André,

No problems at all. We appreciate the great work you are doing.

Thanks

F1ATB, 2021-11-17:
On the bottom of this page,
https://f1atb.fr/index.php/2021/10/22/remote-sdr-v3-manual-installation/
I have put a diagram on the ports.

73
[image: Archiv3.JPG]

closed 2022-03-11 by F1ATB