F1ATB/Remote-SDR #14: Freely tuneable frequency selection

vwvanfan, 2021-12-11:
Hello. Just downloaded the latest version for Orange Pi. Have been looking for ages for a webpage based solution. 
I know this is mainly an amateur radio application, and therefore appreciate the pre defined bands. I'm a theatre sound engineer looking to monitor his wireless microphones. I therefore need 606-614MHz and 850-865MHz. Would there be an opportunity to provide either these bands or a free entry frequency selection? 
Great work. Thank you!

F1ATB, 2021-12-11:
It is very simple. Open the configurationRX.js file and  add 2 bands
corresponding to your need
in the table below line 10
BandesRX.push([606000000,614000000,"My Band 600MHz"]);
BandesRX.push([850000000,865000000,"My Band 850MHz"]);

You can do the same for TX if needed in configurationTX.js file

vwvanfan, 2022-02-10:
Thank You
I'm having a really hard time finding this file. Where might I find it? I'm using the Orange Pi image. 
Thanks

F1ATB, 2022-02-10:
Go here:
/var/www/Html

You will find all the files.
Regards

André
F1ATB

vwvanfan, 2022-02-10:
Excellent, thank you so much. It's working really well. I've now looked closer in the Tools section and realised I could have answered my own question as the link is in there for the file. Thanks for your work

closed 2022-02-10 by vwvanfan