F1ATB/Remote-SDR #16: Problème décodage signaux ACARS avec JAERO

F21467, 2021-12-14:
Bonjour, j'utilise Remote SDR avec un OrangePi , une clé SDR et une antenne RTL-SDR ACTIVE L-BAND PATCH ANTENNA pour recevoir les signaux en bande L AERO d'Inmarsat. Orienté vers le satellite Alphasat, j'obtiens un signal et des datas mais le décodage des messages ACARS ne se fait pas; on voit apparaitre des bad CRC et des Error short frame. Y aurait il des paramètres à régler dans le RX de RemoteSDR (peut être dans GNU Radio) pour essayer de résoudre ce problème
Merci
[image]

F1ATB, 2021-12-14:
Bonjour
Je n'ai pas encore beaucoup d'expérience de réception digital avec Remote
SDR. C'est dans la liste des travaux à venir.
En attendant je vois que vous passez par un câble virtuel. C'est bien.
Néanmoins, votre filtre audio est à 1206HZ pour le passe bas. Il faut le
monter au maximum. Je vois que le spectre ACAR s'étale sur environ 2kHz. De
même il faut descendre la fréquence basse, je ne sais comment agisse
exactement les filtres du navigateur.
Je ne pense pas qu'il faille agir dans GNURADIO, il y a plus de 3kHz de
bande passante.

Comme je travaille actuellemnt sur une refonte du traitement Audio, je vais
rajouter une option pour ne pas passer à travers les filtres passe haut et
passe bas du navigateur. Ce sera disponible dans la version 4.2 qui sortira
d'ici 1 semaine ou 2.

73
André
F1ATB

F1ATB, 2021-12-14:
Autre point
Pensez à régler la dynamique du waterfall (trop rouge) et le spectre.
Flèche/bouton en bas à droite.

F21467, 2021-12-14:
Bonjour,

Merci de votre réponse, je ne suis qu'un "utilisateur / receveur" et je ne
suis pas vraiment au fait des subtilités SDR , je me suis rapproché du
créateur de JAERO qui m'a indiqué

"Looks like the constellation is collapsing. Probably samples being dropped
by the computer or whatever SDR you have set up. Try reducing the SDR
sample rate."

J'ai donc essayé par l'intermédiaire de GnuRadio Companion de faire varier
ce paramètre sans plus de succès, votre option de jouer sur la suppression
des filtres me semble plus prometteuse car un membre d'un groupe Facebook
m'avait conseillé de supprimer tous les filtres mais je ne sais pas comment
faire

Merci de votre aide

Cordialement

R.FERNANDEZ

F1ATB, 2021-12-30:
Je viens de mettre en ligne la version 4.2. Elle dispose d'une sortie RX sans filtrage. Cela devrait mieux fonctionner pour votre décodage numérique.
73
André

F21467, 2021-12-30:
Merci beaucoup, j'essaierai sûrement demain et je vous tiendrais au
courant;
Passez de bonnes fêtes de fin d'année

Cdt
Robert

F21467, 2021-12-30:
je n'ai pas pu m'empêcher d'essayer, j'étais sur OrangePi ,je suis donc passé sur RaspberryPi4 pour cette version mais je n'ai aucun son qui remonte....
[image]

F1ATB, 2021-12-30:
Descendez le squelche au minimum

F21467, 2021-12-31:
Merci , j'ai descendu le squelche au minimun sans effet notable; au niveau
di Pi 4 , y aurait il qque chose à paramétrer ou à rajouter au niveau du
traitement du son. J'ai vu apparaitre sur le desktop Debian de l'Orange PI
(pour qui le son vers le PC fonctionne) le haut parleur avec une croix et à
côté un "carré" pulse audio; faut il plulse audio sur le Pi ? Merci

F1ATB, 2021-12-31:
Non on utilise pas l'audio du PI.

Faites des CTRL-F5 (plusieurs fois si besoin)  au cas ou ce problème serait
dû au cache.
À la vue de votre image. Les échanges se font bien pour le spectre. L'audio
devrait faire de même.

Ouvrez la console du navigateur et CTRL-SHIFT I pour voir les messages
d'erreur.

73

F21467, 2021-12-31:
Bonjour,guère mieux, désolé je ne suis pas très doué, j'ai fait une copie
d'écran des "messages d'erreurs" supposés, mais un élément m'interpelle
toujours, c'est la mention "Input Audio Buffer Size" dans la fenêtre Audio

F21467, 2021-12-31:
Re, je viens d'installer la V4.2 sur l'OrangePI,on diraie le même phénomène

F1ATB, 2021-12-31:
Envoyez-moi une copie de la console, pour comprendre l'erreur.
(CTRL-Majuscule I)

[image: image.png]

F21467, 2022-01-01:
Ctrl Maj I je ne vois rien se passer, j'ai refait un Ctrl Shift I , cela
vous suffit il ? j'ai aussi remarqué que je n'ai aucune action sur le While
RX

F1ATB, 2022-01-01:
A la vue de votre ecran, je vois qu'aucune audio n'arrive.

Qel navigateur utilisez vous? Un CTRL-SHIFT I  (i et non 1 ou L) doit
ouvrir la console. Ou bien vous faite bouton droit, inspecter ppuis onglet
console. On a ensuite une série de messages qui donneront l'erreur.

73
André

F21467, 2022-01-01:
Bonsoir, j'utilise Chrome comme navigateur, ci joint messages "erreurs"

F1ATB, 2022-01-02:
Vous avez oublié de joindre le message d'erreur de la console.
73

F21467, 2022-01-02:
Est ce ceci; sinon donnez moi s'il vous plait la manip exacte à faire,Merci

F1ATB, 2022-01-02:
Avez vous Teamviewer? Ainsi je pourrai prendre le contrôle à distance de
vote PC pour debuger et comprendre ce qui ne va pas.

73
André
F1ATB

F21467, 2022-01-02:
Je viens de l'installer mais jamais utilisé

F21467, 2022-01-02:
Que vous faut il ? Mon ID ?

F1ATB, 2022-01-02:
Envoyez sur mon Email privé et non Github, un ID et le mot de passe que
vous propose teanwiever.

Email :  andre.buhart at gmail.com

F21467, 2022-01-03:
Problème réglé avec l'aide du développeur, Merci beaucoup à lui

closed 2022-01-03 by F21467