F1ATB/Remote-SDR #17: Update/upgrade-error

loewal, 2021-12-30:
Hi,
After apt update & apt upgrade I get the following error:
The following additional packages will be installed:
  libc6
Suggested packages:
  glibc-doc
Recommended packages:
  libnss-nis libnss-nisplus
The following packages will be upgraded:
  libc6
1 upgraded, 0 newly installed, 0 to remove and 95 not upgraded.
6 not fully installed or removed.
Need to get 0 B/2,458 kB of archives.
After this operation, 2,048 B of additional disk space will be used.
Do you want to continue? [Y/n]
Preconfiguring packages ...
(Reading database ... 148642 files and directories currently installed.)
Preparing to unpack .../libc6_2.31-13+deb11u2_arm64.deb ...
ERROR: Your kernel version indicates a revision number
of 255 or greater.  Glibc has a number of built in
assumptions that this revision number is less than 255.
If you\'ve built your own kernel, please make sure that any
custom version numbers are appended to the upstream
kernel number with a dash or some other delimiter.

dpkg: error processing archive /var/cache/apt/archives/libc6_2.31-13+deb11u2_arm64.deb (--unpack):
 new libc6:arm64 package pre-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/libc6_2.31-13+deb11u2_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

And..
apt --fix-broken install didn't fix it...

How to solve please.....

F1ATB, 2021-12-30:
Which processor are you using?
RPI4 Orange Pi One, Zero 2?

Llibc6 is a system library. I have no idea, how to solve this issue.
73
André

loewal, 2021-12-30:
Zero2.

????????

Loe Walter
Insula 103
6416 BZ  Heerlen
☎️ +31-6-24476963

closed 2022-02-09 by F1ATB