F1ATB/Remote-SDR #20: tx with hackrf one unstable

F4IYV, 2022-02-06:
Hi André,

I have strange issue on Remote SDR in TX mode. I use hack RF one for TX and RTL SDR for RX. I run your image on PI4 4.4 (last one)
When I press tx , the system pass in tx (with red key) but after 2 or 3 cycles the Hack RF crash. :(
After that when I go in tool --> hack RF I can't see the hack RF (it's like unpluged) 
If I connect disconect the usb cable to the rf one It's work again for 2-3 TX cycle then crash.

in this test I have disconnect PA and preaplifier to avoid any problem of HF
5V of the pi seems stable 

do you have an Idea?
many thanks
Jeremy

F1ATB, 2022-02-06:
2 possible solutions.
- change the power supply. between 5V or 5.3v, you can have different
situations.
- change the USB cable to the Hack RF. With some cable I had some issues.

F4IYV, 2022-02-08:
Hi André,

I have verify the power supply and cable, same problem also with antoher pi. (I use lab Pwr supply)
on my computer no problem with SDangel.
additional information: the tx led stay off when I press tx.

I run HackRF
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc27a76cc3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x006f4f64

F1ATB, 2022-02-09:
Enter the last 4 digit 6cc3  of the serial number in the
configurationTX.js   file    at the bottom.

closed 2022-02-10 by F1ATB