F1ATB/Remote-SDR #21: Problem with audio.

ombi61, 2022-02-06:
Hi,
I install the Raspberri pi4 image with a RTL SDR as receiver. 
Theremote SDR  (http://192.168.xx.xx seems) to work properly but I no audio is received even if on display (FM band) there are several strong signal,
Any suggestion?
Thanks

F1ATB, 2022-02-06:
Did you set the authorization as explained here:
https://f1atb.fr/index.php/2022/01/06/microphone-and-signal-processing-authorization/

ombi61, 2022-02-06:
You are right!!!!
Now it  works!
Now the problem is that my hackRF box is not viewed by remote-sdr, with
lsusb i find the small R820T2 sdr (RX) but no hackRF...TX..
73

F1ATB, 2022-02-06:
Probably a blocking situation between the HackRF and RPI on the USB. Power off a few second, then On.

F1ATB, 2022-02-06:
You must see the green light (USB)  on the hackrf,

closed 2022-02-10 by F1ATB