F1ATB/Remote-SDR #23: Raspberry Pi image locking up.

Derek000, 2022-02-13:
I have 2 x version 4.4 Raspberry Pi 4B 4Gb installations which seem to have the same issue.
After upgrading from 4.3 to 4.4 image (not just copy content), both devices eventually (10 hours) stop responding on the network (can no longer ping) and the ethernet activity light flashes at a high rate.

Device 1: 1x rtl-sdr
Device 2: 2x HackRf

After reboot both of them recover and operate without issues. 

I haven't done a OS upgrade yet, which may help. 

Thanks
Derek

F1ATB, 2022-02-14:
I have no idea about this issue. Remote SDR don't touch the communication
softwares. At home I have a raspberry 2gb and 2 Hackfr which run h24.

Derek000, 2022-02-14:
Thanks. I'll check the logs in both devices and report anything I find that might contribute and try an update/upgrade.

Thanks again.

F1ATB, 2022-05-14:
Derek
Recently, I had a similar issue with a Raspberry  stop responding on the network. Adding a fan was the solution as described here:
([https://f1atb.fr/index.php/2022/01/06/cpu-cooling/](https://f1atb.fr/index.php/2022/01/06/cpu-cooling/))

closed 2022-06-04 by F1ATB