F1ATB/Remote-SDR #24: changing frequency without mouse

renerrrr, 2022-05-17:
When using a (common) touchpad on a laptop and no mouse, it is very difficult to change the frequency. It is possible to configure a 'scrollfunction' on a laptop but it's not standard. It would be great if there is an alternative to change the frequency.

F1ATB, 2022-05-17:
I agree that with a touchpad, it is not easy to change the frequency. Nevertheless, there is an alternative with the left and right arrow of the keyboard.

F1ATB, 2022-09-03:
You can also construct a rotary knob as described here:
[https://f1atb.fr/index.php/2021/12/06/rotary-knob-and-morse-manipulator-for-remote-sdr/](https://f1atb.fr/index.php/2021/12/06/rotary-knob-and-morse-manipulator-for-remote-sdr/)

F1ATB, 2022-09-11:
In version 5.07, I have intoduced new features to change the frequency. Go on "Tools" page and click on "Look for updates".
73
F1ATB