F1ATB/Remote-SDR #25: Where is   Look for updates

Fihl, 2022-09-03:
I just upgraded to 5.0, and see that front page of web page says: 
> Version 5.02 is installed directly by clicking on "Look for updates" in the "Tools" page.
I have Tools next to Settings, bottom right. 
But I cannot locate any Upgrade possibility
(github also gives this instruction)

Problem I just got, upgraded from v4, burning a new SD from image to my RPi400
- Manual correct always start on -165000Hz
- I cannot select "QO-100 Up" on TX page. It just jumps back to 70cm

I do have RX sound 
I have cleaned Chrome local storage

Thanks for nice system
73 de OZ1AAB

Fihl, 2022-09-03:
PS: I am using Adalm Pluto

F1ATB, 2022-09-03:
First you have to go on the Settings page.
Just below the top, set the Access Key to 1234 

Then you go on the Tools page and on the line 7, you have the look for updates; Click it 2 times.

Explanations and screen pictures are given here;
[https://f1atb.fr/index.php/2022/05/21/remote-sdr-v5-raspberry-4b-or-orange-pi-image-installation/](https://f1atb.fr/index.php/2022/05/21/remote-sdr-v5-raspberry-4b-or-orange-pi-image-installation/)
73
F1ATB

Fihl, 2022-09-03:
Thanks
"Click 2 times" :-) 
My Adalm Pluto RX "man. correction" needs to be around plus 28000, so when changed i settings to be 25000..32000, it now stays as I need
TX is now possible to select, as 1234 is entered i Settings. I have not been online yet though. 
Last fight is that Chrome on pc gives me "continue to unsecure site", while Chrome on Mac does not even though I have activated chrome://flags/#allow-insecure-localhost   Strange..
73 Christen

F1ATB, 2022-09-04:
Christen

Deactivate in chrome " allow-insecure-localhost "
In version 5, i have put an https access with a self-signed certificate.
You don't need to put an exception as in version 4.

Regards
F1ATB

F1ATB, 2022-10-11:
you need to enter the access key '1234' in the top of the setting page.

Regards
73
F1ATB

Fihl, 2022-10-11:
I got the settings to work, using 1234
But only on Chrome on Windows PC

> Deactivate in chrome " allow-insecure-localhost "
On both my macs (M1 and an old MacMini), I cannot get access to the frontpage. 
Sometimes I get 2 seconds of webpage, and then it jumps into "Privacy warnings" **without** a button saying "proceed to unsafe page"

Message says:
172.26.1.118 normally uses encryption to protect your information. When Chrome tried to connect to 172.26.1.118 this time, the website sent back unusual and incorrect credentials. This may happen when an attacker is trying to pretend to be 172.26.1.118 or a Wi-Fi sign-in screen has interrupted the connection. Your information is still secure because Chrome stopped the connection before any data was exchanged.

You cannot visit 172.26.1.118 at the moment because the website sent scrambled credentials that Chrome cannot process. Network errors and attacks are usually temporary, so this page will probably work later.

I have not been much on this problem, as the PC Chrome does work
Christen

F1ATB, 2022-10-11:
- Deactivate in chrome " allow-insecure-localhost ". This was for version 4
and not needed for version 5 of Remote SDR.
- in the Tools page, click on

   - Reset Local Storage in browser

If you do not succeed to go on the Tools page, Rest Manually the Local
Browser.
Ctrl+Shift+I when you are on the page. In Local Storage, clean everything
belonging to Remote SDR.

If you do not succeed, I can take the control of your PC and investigate.

Regards
***@***.***

Fihl, 2022-10-11:
I cannot clear local storage, as the page are not there (as problem page are shown). 
Then I change IP on RPi, and then storage must have gone :-) 
No test-features are enabled. 
I created another user on mac and then got a clean chrome. 
When I start on http://IP (not HTTPS), then I see page shortly, but gets thrown into HTTPS within 2 seconds, and then the complain. 

Nothing helps. Must be some certificate problem. 
You can look into my Mac for sure. Catch me on Christen@Fihl.net or OZ1AAB@Fihl.net 
I am away 2:30-3:00

F1ATB, 2022-10-11:
> You can look into my Mac for sure. Catch me on ***@***.*** or

You must accept the certificate. It's a self-certified certificate create
by me. It's free. You accept validate as explained here
https://f1atb.fr/index.php/2022/05/21/remote-sdr-v5-raspberry-4b-or-orange-pi-image-installation/

and you will switch to https which is mandatory to access the microphone on
Chrome.

Fihl, 2022-10-11:
I must give up on Chrome on Mac

Safari and Firefox can browse to http://remsdr-rpi4.local/ 
Both asks "Are you damn sure", but lets me continue to site, and can listen. 

In Chrome I have tried a lot, and "Insecure content" set to Allow on this page: 
chrome://settings/content/siteDetails?site=https%3A%2F%2Fremsdr-rpi4.local

But I never gets a button saying "Continue to site anyway". 
Sorry

Fihl, 2023-05-08:
I am back, on QO, and started over. 
Got a V5.0 on RPi4. It works on pc and now also (again) on Mac too
Upgraded it to V5.08, and it also works perfekt. 
I put the previous SD card back (V5.06 I guess, maybe V5.02). No luck

Strange though, 
but I am now on the air, on V5.08

Case closed, with success, thanks
73