F1ATB/Remote-SDR #26: CAT issue

AndyBlm, 2022-10-17:
Hi Andre!
I leave a message on you site but today my post disappered 
Nevetheless, i have installed rpI4 image and updated to 5.07 and still have issue with CAT
GPREDICT and setting FREQA via Omnirig Client works fine, however OmniRig did`t retrieving freq from RemSDR - field FREQ  in client apps always empty and in  outgoing tab of VSPE i can see only IF5 or IF7 and nothing else
What is wrong ? pls help

F1ATB, 2022-10-17:
I answered this morning.
https://f1atb.fr/index.php/2022/01/18/omnirig-remote-sdr-2/

AndyBlm, 2022-10-17:
Andre, thank you for reply, i attached a few screenshots and hope will be clear for you 
every time after start OmniRig client i see the new message in outgoing tab of VSPE:  IF0 IF1 etc
but no frequency are returned  
[image]
[image]
[image]
[image]
[image]

F1ATB, 2022-10-17:
Andy

Which software (YONIQ-SSTV, WSJT-X...) is connected to Omnirig? It must
provides the frequency.
André
***@***.***

AndyBlm, 2022-10-17:
Andre, for the time being i trying to use Log only without SSTV or WSJT apps , i hope receiving Freq should return to Log or ?
With setting freq via field FREQA on OmniRig Client i can see FA000435600000 command in VSPE incoming messages  however in outgoing no anything different in compare with my pics in the post above.

F1ATB, 2022-10-17:
You need to transmit continuously the frequency. Otherwise, there is a
timeout and you are disconnected.

André

AndyBlm, 2022-10-17:
Can be a turnaround just to retrieve freq without continiously send send send?
BTW in you article -Test with “Client.exe”- visible retrived freq in the tab FREQ what i can`t get
[image]