F1ATB/Remote-SDR #29: orange pi 4 lts support

zochowsd, 2022-12-12:
I have orange pi 4 lts.... i installed all components with manual procedure..
so i need to make adjustments in CPU.js. :).
What its interesting this CPU has 6 cores with 4 type a53 and 2 type a72

should  i based on a72 or a53? or both ? 

lscpu - printout:lscpu 
Architecture:           aarch64
  CPU op-mode(s):       32-bit, 64-bit
  Byte Order:           Little Endian
CPU(s):                 6
  On-line CPU(s) list:  0-5
Vendor ID:              ARM
  Model name:           Cortex-A53
    Model:              4
    Thread(s) per core: 1
    Core(s) per socket: 4
    Socket(s):          1
    Stepping:           r0p4
    CPU max MHz:        1416.0000
    CPU min MHz:        408.0000
    BogoMIPS:           48.00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
  Model name:           Cortex-A72
    Model:              2
    Thread(s) per core: 1
    Core(s) per socket: 2
    Socket(s):          1
    Stepping:           r0p2
    CPU max MHz:        1800.0000
    CPU min MHz:        408.0000
    BogoMIPS:           48.00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
NUMA:                   
  NUMA node(s):         1
  NUMA node0 CPU(s):    0-5
Vulnerabilities:        
  Itlb multihit:        Not affected
  L1tf:                 Not affected
  Mds:                  Not affected
  Meltdown:             Not affected
  Mmio stale data:      Not affected
  Retbleed:             Not affected
  Spec store bypass:    Vulnerable
  Spectre v1:           Mitigation; __user pointer sanitization
  Spectre v2:           Vulnerable
  Srbds:                Not affected
  Tsx async abort:      Not affected

zochowsd, 2022-12-12:
root@opi4lts-ham:/remsdr# node Radio_Server.js 
node:internal/fs/utils:347
    throw err;
    ^

Error: EINVAL: invalid argument, write
    at Object.writeSync (node:fs:920:3)
    at Object.writeFileSync (node:fs:2208:26)
    at exportGpio (/remsdr/node_modules/onoff/onoff.js:18:8)
    at new Gpio (/remsdr/node_modules/onoff/onoff.js:172:36)
    at Object.<anonymous> (/remsdr/remsdr_modules/CPU.js:44:17)
    at Module._compile (node:internal/modules/cjs/loader:1155:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1209:10)
    at Module.load (node:internal/modules/cjs/loader:1033:32)
    at Function.Module._load (node:internal/modules/cjs/loader:868:12)
    at Module.require (node:internal/modules/cjs/loader:1057:19) {
  errno: -22,
  syscall: 'write',
  code: 'EINVAL'
}

F1ATB, 2022-12-12:
Honestly, I don't know. I have not yet used an Orange Pi 4 LTS