F1ATB/Remote-SDR #31: Disable TX Lo

Sergey-R5AU, 2023-02-27:
Hi André !
Thank you  for nice programming!
Works fine CW, DIGI and SSB QSOs i did via QO-100
Could you modify your code with disabling TX LO of AD93XX  within simplex mode ?
Othervise TX LO disturbing normal reception and plenty of carriers are visible especially on high frequencies , see attached pic
Thank you  in advance !
[image]

F1ATB, 2023-02-27:
I have to investigate the feasibility as it takes time to set the parameters of the transmitter when you want to transmit.  For the moment, I put the transmit gain to the minimum. It's why I don't understand why you have all these lines. 

73
André

Sergey-R5AU, 2023-02-27:
Andre thank you for quick reply !
Well , as i mentioned this is started visible with frequency above 500-600 mhz, were on 2m or 70cm practically thismis not disturbing too much.
However i learned a 2 wayaround:
1 - disable TX LO during reception - simple simplex mode
2 - move TX frequency out of RX range (ex.: 1mhz or more below or up - aka SPLIT mode)