F1ATB/Remote-SDR #34: Supported browsers?

sh123, 2023-04-23:
Running Remote-SDR on RPI 4 with Bullseye distro, running Debian Bullseye on PC.

Receive works, but when trying to transmit:
- No errors from nodejs console
- Firefox shows "SDR TX On" with green, microphone is selected, clicking on "TX audio on" makes button red, but no TX signal is shown on spectrogram. Firerefox does not pick any mic audio?
[image]
- Trying with latest chrome, "SDR TX On" is shown with blue, clicking on "TX audio" button does not turn it red, mic signal is visible on spectrogram, so it picks up mic audio, but cannot start transmission
[image]
- Also in chrome "select frequency band" does not work, old band selections is selected back when trying to select new band
- Transmit works on RPI4 with, e.g. SDRAngel and command line tools

F1ATB, 2023-04-24:
As mentioned in the documentation, don't use Firefox. Use Chrome or Edge browsers.
If you have still soe issues. Go to the troubleshooting page :
https://f1atb.fr/index.php/2022/02/15/troubleshooting/
You will find different issues and haw to solve them.
Regards 73
André F1ATB

sh123, 2023-04-24:
Unfortunately, none of FAQ items helped, anyway, I'm closing this item for now.

closed 2023-04-24 by sh123