F1ATB/Remote-SDR #36: New version, Update etc.

Sergey-R5AU, 2023-10-17:
Hi André !
I hope you well !
sine 5.08 release plenty of SW updates already release and btw looks like some new features HAM`s are looking for.
Could an updates of Remote SDR are expectable in 2023 ?

F1ATB, 2023-10-17:
May be this winter. Lack of time.

Regards

Sergey-R5AU, 2023-10-17:
Thank you  Andre !
Appreciated !

In my shack R-SDR in use with all updates of RPI OS in 64bit mode, modification of CPU.js was required:
instead of  "if (info_cpu.indexOf("armv7l") > 0)"
use:  "if (info_cpu.indexOf("aarch64") > 0)" 
due to already   uname -a as:
17:24:16 BST 2023 **aarch64** GNU/Linux

and need to update shell file for PM2:

sudo nano /etc/init.d/pm2-init.sh

instead of /root/.pm2   should be   /home/pi/.pm2   

in few strings  otherwise PM2 service not started