F1ATB/Remote-SDR #40: Lost ability manage GPIO pin with TX/RX state change

Sergey-R5AU, 2023-11-27:
Hi Andre !

after some updates of Pluto i lost ability to driving GPIO with changing RX/TX state with:
[image]
[image]

I have tested separately onoff package with simple JS script like:

'use strict';
const Gpio = require('onoff').Gpio;  
const led = new Gpio(6, 'out');  
led.writeSync(1);     

GPIO6 driving fine as previously , but state not changing with TX/RX RemSDR  - with **TX Audio** button or with external switch based on arduino. 

What could be wrong, what i can trace and check ?

Thank you !

Sergey-R5AU, 2024-02-01:
I have been spending a time for isolation and solving problem above.
Even on the last version completely lost TX audio !
On the fact:  core of issue is -  version of the Browsers ,  Chrome or Edge.
I roll back to the version of the Chrome 110.xx - all works fine.
Problem:  new version are not supporting used node.js and especially incompatible with used version of socket.io
Recommended to use latest version of node.js and t libraries.
Problem solved.

Sergey-R5AU, 2024-02-01:
Problem solved as mentioned in the last post. Latest versions of browsers are not supporting used version of node.js and socket.io with used in RemSDR images.

closed 2024-02-01 by Sergey-R5AU

dona76, 2024-02-21:
Very nice software André !
I've fixed the TX issue with rolling back Chrome to 104 version (ether 32 or 64b) and following method 1 in 
https://www.browserstack.com/guide/downgrade-to-older-versions-of-chrome
With Group Policy, it is possible to disable gogole update.
73' F4BSP