F1ATB/Remote-SDR #44: Image for RaspberryPi 5

audric, 2024-06-12:
Hello, any chance to have a working image for the RaspberryPi 5 ? Or at least a workaround.

The RaspberryPi 4 doesn't boot. It complains about not being the right architecture.

compugician, 2024-09-13:
I'm trying to set this up on a Pi 5, Once it works I'll share.

I have it mostly working - but for some reason the XMLRPC calls to set RX/TX parameters are not having any effect. 

There are some comments in the code about this and I'm trying to figure the whole thing out.

If anyone has any experience with this please share :-)

neffen37, 2025-02-03:
hello I would also be very interested in a functional version on the pi5 because unfortunately the other websdr alternatives are not compatible with the pi5 either.

merci d'avance

F1ATB, 2025-02-04:
Understand, but for the moment I am busy on other subjects.

Regards

André