F1ATB/Remote-SDR #46: Port forwarding

Harrie1966, 2024-10-07:
I have a question? Is it possible that you can also view this web remote outside your own network? using port forwarding or something. Who can help me further? Thank you in advance.

F1ATB, 2024-10-08:
Yes you can forward the port as described here :

https://f1atb.fr/communication-ports/

73
André

Harrie1966, 2024-10-11:
I have done a port forwarding 443. And now I can see the website. but cannot adjust anything, he does not see the dongle and cannot do anything. What am I doing wrong?
[image]
[image]

F1ATB, 2024-10-11:
Ctrl-Shift I to open the browser console and analyse the error messages
73
André

Harrie1966, 2024-10-12:
I solved the problem by installing a VPN server. Tailscale this worked well straight away. I think KPN in the Netherlands does not allow port 443. So now it works fine. now another question, sdr-rtl dongle version 4, will it also work in this version of yours?